dirkhh/adsb-feeder-image #246: Request key for FlightAware fails silently

matburnham, 2024-10-24:
### Issue

Request Key for FlightAware fails silently.

### Steps to recreate

1. From a new install, go to Data Sharing page.
2. Select FlightAware
3. Hit Request Key
4. Page returns with no value filled in.

Logs:

```
2024-10-24T18:22:21.233Z handling flightaware--key -> ''
2024-10-24T18:22:21.233Z handling flightaware--submit -> stay
2024-10-24T18:22:21.233Z got aggregator object <utils.other_aggregators.FlightAware object at 0x7f800f1110> -- activating for sitenum 0
2024-10-24T18:22:21.235Z download_docker_container ghcr.io/sdr-enthusiasts/docker-piaware:latest-build-544
Error response from daemon: Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
2024-10-24T18:23:45.149Z docker run --rm ghcr.io/sdr-enthusiasts/docker-piaware:latest-build-544 received a timeout error after 45.0 with output b'INFO: no specific FEEDER_ID set, falling back to dynamic generated (/var/cache/piaware)\nWARNING: TZ environment variable not set\nallow-auto-updates "no"\nallow-manual-updates "no"\nallow-mlat "yes"\nmlat-results "yes"\nreceiver-type "rtlsdr"\n[skyaware] 2024/10/24 18:23:28 2024-10-24 18:23:28: (server.c.1704) server started (lighttpd/1.4.69)\n[piaware] 2024/10/24 18:23:30 ****************************************************\n[piaware] 2024/10/24 18:23:30 piaware version 9.0.1 is running, process ID 133\n[piaware] 2024/10/24 18:23:30 your system info is: Linux 8fefc6534090 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux\n[piaware] 2024/10/24 18:23:32 Connecting to FlightAware adept server at piaware.flightaware.com/1200\n[piaware] 2024/10/24 18:23:32 Connection with adept server at piaware.flightaware.com/1200 established\n[piaware] 2024/10/24 18:23:32 TLS handshake with adept server at piaware.flightaware.com/1200 completed\n[piaware] 2024/10/24 18:23:32 FlightAware server certificate validated\n[piaware] 2024/10/24 18:23:32 encrypted session established with FlightAware\n[piaware] 2024/10/24 18:23:33 logged in to FlightAware as user guest\n[piaware] 2024/10/24 18:23:33 my feeder ID is redacted-dead-beef-dead-beefdeadbeef\n[piaware] 2024/10/24 18:23:33 no ADS-B data program seen listening on port 30005 for 3 seconds, next check in 60s\n[piaware] 2024/10/24 18:23:33 UAT support disabled by local configuration setting: uat-receiver-type\n'
2024-10-24T18:23:55.161Z error activating flightaware--submit: cannot access local variable 'result' where it is not associated with a value
2024-10-24T18:23:55.161Z did not successfully enable flightaware
```

Breaking out the Docker errors with newlines broken out:

```
docker run --rm ghcr.io/sdr-enthusiasts/docker-piaware:latest-build-544 received a timeout error after 45.0 with output b'INFO: no specific FEEDER_ID set, falling back to dynamic generated (/var/cache/piaware)
WARNING: TZ environment variable not set
allow-auto-updates "no"
allow-manual-updates "no"
allow-mlat "yes"
mlat-results "yes"
receiver-type "rtlsdr"
[skyaware] 2024/10/24 18:23:28 2024-10-24 18:23:28: (server.c.1704) server started (lighttpd/1.4.69)
[piaware] 2024/10/24 18:23:30 ****************************************************
[piaware] 2024/10/24 18:23:30 piaware version 9.0.1 is running, process ID 133
[piaware] 2024/10/24 18:23:30 your system info is: Linux 8fefc6534090 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
[piaware] 2024/10/24 18:23:32 Connecting to FlightAware adept server at piaware.flightaware.com/1200
[piaware] 2024/10/24 18:23:32 Connection with adept server at piaware.flightaware.com/1200 established
[piaware] 2024/10/24 18:23:32 TLS handshake with adept server at piaware.flightaware.com/1200 completed
[piaware] 2024/10/24 18:23:32 FlightAware server certificate validated
[piaware] 2024/10/24 18:23:32 encrypted session established with FlightAware
[piaware] 2024/10/24 18:23:33 logged in to FlightAware as user guest
[piaware] 2024/10/24 18:23:33 my feeder ID is redacted-dead-beef-dead-beefdeadbeef
[piaware] 2024/10/24 18:23:33 no ADS-B data program seen listening on port 30005 for 3 seconds, next check in 60s
[piaware] 2024/10/24 18:23:33 UAT support disabled by local configuration setting: uat-receiver-type
'
2024-10-24T18:23:55.161Z error activating flightaware--submit: cannot access local variable 'result' where it is not associated with a value
2024-10-24T18:23:55.161Z did not successfully enable flightaware
```

Not sure if the issue is in this repository or https://github.com/sdr-enthusiasts/docker-piaware, but it looks like docker-piaware is doing it's thing, but the result isn't getting to adsb-feeder-image?

### Workaround

Manually copy the key from the log to the key box for FlightAware.

wiedehopf, 2024-10-24:
docker-piaware log format changed.

wiedehopf, 2024-10-24:
I suppose it could be from this line:
```py
            except subprocess.TimeoutExpired:
                print_err(f"failed to remove the temp container {str(result.stdout)} / {str(result.stderr)}")
```
Yeah it might have failed to remove the container which would seem to be destined to cause an exception.

wiedehopf, 2024-10-24:
Yeah pretty sure it's a timeout there.

```py
print_err(f"failed to remove the container {name} stderr: {str(exc2.stdout)} / stdout: {str(exc2.stderr)}")
```
This likely fixes the code but really what you ran into sounds like your system has other issues.

matburnham, 2024-10-24:
Would that error have ended up in my log anywhere, or does it not persist?

wiedehopf, 2024-10-24:
Oh and your system not responding to `docker rm -f temp_container` within 10 seconds just means ... docker isn't happy.
Which could be a resource issue, sd-card issue ... hard to say.

I've made the code more robust, so your point 1 is fixed.
Point 2 as in reporting the error is UI stuff ... yeah maybe i'll open a new issue for that.

closed 2024-10-24 by wiedehopf

wiedehopf, 2024-10-27:
So these fixes are now in beta, if you have any issues with activation in the future just open a new issue please.