dirkhh/adsb-feeder-image #255: Defining a domain

kjorda, 2024-12-03:
Really like this project!  I have gone the VM route and it works great.  However, I am presenting the VM through a proxy am getting the following error when I connect to it through a domain (versus IP address):

`The external IP of your browser and the feeder are different. The information in the status links for some of the aggregators below may be incorrect.`

The error is indeed correct - when I access it from outside my network my browser is seeing my external IP, not the internal VM IP address.  However, I don't see why this should break things.  Can we add a URL name somewhere so that, even if the IP addresses don't match I can still access the maps?

wiedehopf, 2024-12-03:
That's not what this warning is about.

The map is on a different port, that's why your proxy isn't working.
You'll have to make your own URL for the proxy on the map.
Which you then can't click ofc cause it's pointing at a port.

wiedehopf, 2024-12-03:
You can just hover the link and it'll show you the port :)

kjorda, 2024-12-03:
Thanks for the response!  It wouldn't show me the port on hover.  Just my domain and then the folders /maps and /stats.  No worries though.  I am using Zoraxy and just created the two following virtual directory rules and got what I was looking for:  
  
`/map/ => <internal ip>:1090`  
`/stats/ => <internal ip>:1090/graphs1090/`  
  
The stats link on the map takes me to /map/graphs1090/ and that works too so I think I'm happy.  
  
An honor to message with you, @wiedehopf.  Have been using and enjoying your code for years.

closed 2024-12-03 by kjorda