dirkhh/adsb-feeder-image #257: login not working

edwinov, 2024-12-08:
Images tried:

adsb-im-x86-64-vm-v2.2.4-Hyper-V-x86_64.vhdx.xz
adsb-im-x86-64-vm-v2.2.4-VirtualBox-x86_64.ova.xz

After booting up the VM it comes up with

`- LAN IP : 192.168.2.130

Default Login:
Username = root
Password = dietpi (or custom dietpi.txt entry)

Please hit <return> to login`

Ok, so I try to log in using exactly what is suggested : root/dietpi but it comes back with 'Login incorrect'. Same when I try to ssh into it.

Can anyone tell me what the password is supposed to be and shouldn't this be added to the documentation? I can't be the only one having trouble with this?

Thanks.

wiedehopf, 2024-12-08:
Use the webinterface to set a password.

dirkhh, 2024-12-08:
The images have quasi random passwords. 
You can set a different password in the web UI (or, preferably, set an ssh key)

edwinov, 2024-12-08:
I'm running the image on a server that already has port 80 assigned so I can't visit the web ui, unless it listens to an alternative port as well, does it?

edwinov, 2024-12-08:
Hold on, that doesn't make sense. Sorry, I was confusing my current problem with another one. The problem with trying to connect to the web UI ui is that I can a 'This site can't be reached, x.x.x.x refused to connect'. I can ping it ok though.

But still why does the login screen suggest root/dietpi when it's quasi random?

wiedehopf, 2024-12-08:
It's the default login screen.
Primarily this image is aimed at people who don't know what a VM or console is.

Hard to know why you can't connect to the webinterface but it's likely an issue with your firewall / VM settings?

edwinov, 2024-12-08:
I am running other VMs in it as well (Windows 10, Ubuntu, etc.) and they all connect fine. At a bit of a loss here.

edwinov, 2024-12-08:
Iv'e tried with disabling firewall, no change. In Hyper-V I've tried different Virtual switches for the Network Adapter, no change.

wiedehopf, 2024-12-08:
Possibly the image is just broken and has errors on first boot.
(i really don't know how to work virtualbox well but it gives me some error on first boot)

Maybe just use a small debian base image and then use the app install.

edwinov, 2024-12-08:
There's a mention somewhere that on frist boot there will be an error but on second boot it's fine, and that is indeed the case. So that's not a problem. It's the web ui that's unresponsive. 

Do you have a link to this app install? Not sure what you mean as I've only seen images available for download but if there's some apt-get install method that could work.

wiedehopf, 2024-12-08:
It's on the main page.

Which note are you speaking of?
dietpi has some firstrun stuff that needs to run, it seems that it's trying to do wifi which is not appropriate for the VM images.

edwinov, 2024-12-08:
Ah, yeah, found it on main page, will give that a try. Don't remember where I saw the note. Just did a port scan on the VM image, it's **only** listening on port 22, no http(s) ports at all. Strange.

edwinov, 2024-12-08:
Not having much luck.

Installed latest ubuntu in a Docker container. Ran `curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash`, installed all missing dependencies but now the script keeps coming up with

You appear to be on a debian-style destribution
Please install the missing packages before re-running this script:
apt-get install -y docker-compose

But all packages **are** installed and if I follow the suggestion anyway I get 

apt-get install -y docker-compose
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
docker-compose is already the newest version (1.29.2-6ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

So that's a dead end as well.

wiedehopf, 2024-12-08:
`hash -r` then try again maybe.

```
which docker
docker compose version
docker-compose version
```
what is the output for these commands?

edwinov, 2024-12-08:
#which docker
/usr/bin/docker

#docker compose version
docker: 'compose' is not a docker command.

#docker-compose version
docker: 'compose' is not a docker command.
See 'docker --help'
root@2aabff13564a:/# docker-compose version
Traceback (most recent call last):
  File "/usr/bin/docker-compose", line 33, in <module>
    sys.exit(load_entry_point('docker-compose==1.29.2', 'console_scripts', 'docker-compose')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/docker-compose", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 9, in <module>
    from distutils.spawn import find_executable
ModuleNotFoundError: No module named 'distutils'

wiedehopf, 2024-12-08:
I don't even know how docker is broken on a fresh install .. good stuff ubuntu.

You can just run this script to install a new docker.
It's a VM anyhow:
https://get.docker.com/

wiedehopf, 2024-12-08:
Possibly you could try the 2.2.3 virtualbox image ... maybe 2.2.4 broke the image.

edwinov, 2024-12-08:
Just to be clear: I am running Docker on a server running Windows. I have now tried with Debian, Ubuntu and latest Fedora and none of them play nice with `curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash`

Fedora:

[root@7e31a3287c01 /]# curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  6356  100  6356    0     0  19006      0 --:--:-- --:--:-- --:--:-- 19029
You appear to be on a fedora-style distribution
Please install the missing packages before re-running this script:
dnf install -y python3 python3-flask python3-requests git docker docker-compose

So I do what's suggested:

[root@7e31a3287c01 /]# dnf install -y python3 python3-flask python3-requests git docker docker-compose
Updating and loading repositories:
 Fedora 41 - x86_64 - Updates                                                                                                                                                                  100% |   4.5 MiB/s |   6.3 MiB |  00m01s
 Fedora 41 openh264 (From Cisco) - x86_64                                                                                                                                                      100% |   3.3 KiB/s |   4.8 KiB |  00m01s
 Fedora 41 - x86_64                                                                                                                                                                            100% |   7.4 MiB/s |  35.4 MiB |  00m05s
Repositories loaded.

................
then a whole lot of downloading and installing followed by
...............

Complete!

So I try again:

[root@7e31a3287c01 /]# curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  6356  100  6356    0     0  99225      0 --:--:-- --:--:-- --:--:--   98k
You appear to be on a fedora-style distribution
Please install the missing packages before re-running this script:
dnf install -y python3 python3-flask python3-requests git docker docker-compose

Same message as before : install missing packages, and if I try to it says nothing to do:

[root@7e31a3287c01 /]# dnf install -y python3 python3-flask python3-requests git docker docker-compose
Updating and loading repositories:
Repositories loaded.
Package "python3-3.13.0-1.fc41.x86_64" is already installed.
Package "python3-flask-1:3.0.3-6.fc41.noarch" is already installed.
Package "python3-requests-2.32.3-3.fc41.noarch" is already installed.
Package "git-2.47.1-1.fc41.x86_64" is already installed.
Package "moby-engine-27.3.1-2.fc41.x86_64" is already installed.
Package "docker-compose-2.30.3-1.fc41.x86_64" is already installed.

Nothing to do.

So it's a catch 22.

None of the VM images work for me, neither does the installation in linux as an app. ;-(

wiedehopf, 2024-12-08:
So for the app install your docker compose / docker-compose is just broken.
That's outside adsb.im control. I suggested a potential fix installing docker directly from docker instead from the distribution.
Try that.

I've already mentioned that the 2.2.4 vm images are possibly broken (and i'm looking at that).
Try a 2.2.3 image.

You're running stuff that's less tested that the typical stuff which is mostly images for raspberry pi .. and you had some bad luck.

dirkhh, 2024-12-08:
Fedora hasn't worked in a while
Decision and Ubuntu get reasonably frequently tested. 

I did a Ubuntu install just a few weeks ago. 

You do have to be on a current version. No 20.04-LTS or things like that 

/D

dirkhh, 2024-12-08:
Oh, I misread

No, you cannot run Linux under docker on Windows and expect this to work 
It doesn't

It needs to be an actual VM running Ubuntu or Debian

edwinov, 2024-12-08:
What a stroke of bad luck I'm having. Thought I'd use an actual VM then, not a Docker container. Still had recent Linux Mint .iso on my machine so tried to install that in Hyper-V, no luck, they don't seem to work together. Now downloading Ubuntu .iso which is a whopping 5.8GB at 900KB/s, which will take 2 hours.  Will update progress in this thread.

wiedehopf, 2024-12-09:
2.2.3 image? in releases?

edwinov, 2024-12-09:
Ok, whilst Ubuntu .iso is still downloading I tried with 2.2.3 image (adsb-im-x86-64-vm-v2.2.3-Hyper-V-x86_64.vhdx), same problem as before, site can't be reached. It's only open port is 22, no 80, no 8080, no 443.

[image]
[image]
[image]

dirkhh, 2024-12-09:
Try 1099... But it should be 80

edwinov, 2024-12-09:
Tried that as well, no luck.

edwinov, 2024-12-09:
Meanwhile, trying to run `curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash` in a Linux Mint Virtual Box VM does nothing, that is: it's been displaying this very screen for the past 20 minutes:

[image]

Absolutely nothing seems to be working as expected.

edwinov, 2024-12-09:
Ha! Finally some success. In a Linux Mint VM I finally got the web server to play ball:

[image]

wiedehopf, 2024-12-09:
So the 2.2.4 virtualbox ova has been updated.
It should be a much slimmer VM than that linux mint you got going.

It works in my local testing in virtualbox.
The only relevant setting for me was this one:

[image]

wiedehopf, 2024-12-09:
Seems you need to specifically pass through USB

[image]

Also sorry for the VM images being broken, seems 2.2.3 was broken as well i didn't realize how long ago that change was made.
Shows that not that many people use those images as we haven't had other bug reports.

Would be appreciated if you could test the 2.2.4 ova, should give you a much slimmer VM than that linux mint and give us a data point that the image is working not just for me :)

wiedehopf, 2024-12-09:
Ok it seems you have to change the USB settings differently so they aren't temporary.

[image]

Probably like this?

wiedehopf, 2024-12-09:
Possibly even make the filter not depend on the port and a bunch of other details:

[image]

edwinov, 2024-12-09:
New 2.2.4 images for both VirtualBox and Hyper-V working like a charm, using the same USB pass-through as I did with the full fledged Linux Mint VM:

[image]

Thanks for your quick help in getting this sorted, much appreciated.

Mfg,

edwinov

closed 2024-12-09 by wiedehopf