dirkhh/adsb-feeder-image #258: Hyper-V VM not working

edwinov, 2024-12-23:
Hi,

I can't get the Hyper-V image (adsb-im-x86-64-vm-v2.2.4-Hyper-V-x86_64.vhdx.xz) to work because Hyper-V has _no_ USB pass-through. I had no problem getting the VirtualBox image (adsb-im-x86-64-vm-v2.2.4-VirtualBox-x86_64.ova.xz) to work because VirtualBox _does_ have USB pass-through.

So I am wondering if I'm missing something? I mean, why else would there be a Hyper-V image if it can't work to begin with?

I can always use the VirtualBox version if need be but prefer to use Hyper-V.

Cheers/Mfg,

Edwin

dirkhh, 2024-12-23:
I consider running a feeder in a VM a strange corner case. Doing USB is a VM comes with all kinds of issues and the system that runs the VM is unlikely to be conveniently located close to where you want your antenna to be. 
A stage 2 (which doesn't want or need USB) is a much better for for a VM

wiedehopf, 2024-12-23:
see also: https://github.com/wiedehopf/adsb-wiki/wiki/mlat-in-VMs

edwinov, 2024-12-23:
At my gliding club we decided we wanted to contribute to the ADSB community (that we use for tracking of our gliders) but wanted to keep cost low so got just an antenna and dongle and will run it on our club's server that's always on anyway and is physically located close to the antenna, this way there's no need for an extra RPi-like device. 

I've tested the (VirtualBox) image at home (on my own workstation) for the past week and the results were great. Today I decided to move it to our actual gliding club and use Hyper-V. That's when I realised it wouldn't work because of the missing USB pass-through, which VirtualBox _does_ have and has shown to work without any problems. 

So I don't consider it a strange corner case at all. If you already have a server, run it on there. I am just wondering why, especially if you consider it such a strange corner case and even more especially given the fact that Hyper-V doesn't have USB pass-through, is there even an image for it?

Not sure what you mean by 'A stage 2 (which doesn't want or need USB) is a much better for for a VM'. I _do_ need USB as my dongle (rtl-sdr) is USB.

wiedehopf, 2024-12-23:
stage 2 is a use-case of having multiple remote receivers and having them feed / show data / collect stats on a central server.
That's the main usecase for all of the VM images.
You asked why the image exists, that's the answer.
It'll continue to exist even if it can't be used with USB.

USB passthrough in most cases causes MLAT to not work.
If you don't care for MLAT or you could make it work with a VM somehow, good on you.
Doesn't change that we wouldn't recommend it.

edwinov, 2024-12-23:
Ok, I understand, thanks for explaining!

closed 2024-12-23 by wiedehopf