dirkhh/adsb-feeder-image #265: Adding/Changing things on the Stats page?

jason0x21, 2025-01-24:
I'm trying to figure out how to do two things...

1) Change the range distance to metric instead of nautical miles.
2) Add graphs to the state page.

It looks like the image either pulls in @wiedehopf 's graphs1090 code or forks it? But I can't seem to figure out how to change the units.  I've looked at some of the GRAPHS1090_* environment variables, but a search of the code (in both this and weidehopf's repos) turns up only where they're _set_ and not how they're _used_.  I figure changing the range units is the first step on path to adding graphs, but I'm kinda stumped as to how it fits together.  What am I missing here?

wiedehopf, 2025-01-24:
You gotta look at the https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder readme.

You can't add custom graphs.

jason0x21, 2025-01-24:
Got it, thanks!

So if I made contributions graphs1090 to add graphs (and they were accepted, of course) would that be picked up through the various packages?

jason0x21, 2025-01-24:
Also I looked through that and the only thing I could find was TAR1090_DISPLAYUNITS, which I set to 'metric'.  That didn't seem to work once I hit "Submit" and everything came back online (or does it take a while? Like, a longer while beyond just rebooting and feeding again?  Thanks!

wiedehopf, 2025-01-24:
Adding custom graph support to graphs1090 is such a PITA ... i might add one configurable extra temperature graph but that's about it.

Maybe just running https://sdr-enthusiasts.gitbook.io/ads-b with your own fork of ultrafeeder which in turn uses your fork of graphs1090 might be the way?
I'm not saying i won't accept a PR. But i'm also not looking forward to it :)

jason0x21, 2025-01-24:
> 
> Maybe just running https://sdr-enthusiasts.gitbook.io/ads-b with your own fork of ultrafeeder which in turn uses your fork of graphs1090 might be the way? I'm not saying i won't accept a PR. But i'm also not looking forward to it :)

Ha! Yeah, I was looking to add an extra temp graph that uses a (fairly common?) USB temperature sensor to get the ambient temperature of the space the Pi is in.  So it might be a niche to me thing.

I tried a private window but no luck.

wiedehopf, 2025-01-24:
[image]

works fine for me.

yeah extra temp graph has been a common request, so with a bit of luck i'll add one.

jason0x21, 2025-01-24:
Yep, that's what I have, but I'm still not seeing that show up on the page with all the graphs.  Hmm.

jason0x21, 2025-01-24:
Figured it out.  The env var I wanted was actually. GRAPHS1090_RANGEUNITS

closed 2025-01-25 by dirkhh