dirkhh/adsb-feeder-image #267: Replicate HeyWhat'sThat config in a Stage2 feeder to the respective micro feeder

gabo03, 2025-01-26:
I see that my micro feeders don't get the right HeyWhat'sThat configuration from the stage2 feeder.

I can see the silhouettes correctly in the stage2 combined map, but not in every micro feeder that I have (currently 2).

I don't know if this is something missing from my part, but I already change the HeyWhat'sThat configuration to one of my micro feeder because a location change, and I detect that.

Thanks...

wiedehopf, 2025-01-26:
Maps -> "station_name" -> Options

You can most certainly set the hwt id.
And changing stuff on the stage2 is not gonna affect the microfeeder, just the stage2 / stage2 microsite.
The stage2 doesn't control the receiver devices, it just gets data + their location from them.

If you are on the microfeeder, just click Map -> options. 99% sure that works just fine unless you set it to nanofeeder.
I don't really know the goal of this.

gabo03, 2025-01-26:
In both micro feeders going to Map -> options doesn't show the field for hwt id, that's why I thought the stage2 control what micro feeders are showing.

dirkhh, 2025-01-27:
yeah, we don't offer that on the micro feeders. Maybe that's a mistake.
Should be easy enough to change for the next version.

gabo03, 2025-01-27:
Just FYI, these micro feeders were normal feeders before and they keep the HWT configuration for some reason even after I convert them to micro feeders.

dirkhh, 2025-01-27:
Yep. We are just hiding the UI.
Will be fixed in the next beta

dirkhh, 2025-01-27:
It's in the current beta

closed 2025-01-27 by dirkhh