dirkhh/adsb-feeder-image #270: 2.2.6 VM dies after a few minutes

tpvainio, 2025-02-02:
VM (adsb-im-x86-64-native) dies after a few minutes.  Had to return back to 2.2.5 by reinstalling and returning taken backup.  I've been running these few weeks and originally installed it as 2.2.2.  Since then upgraded every time a new stable was available.  This is a VM stage2 install which fails but my 2.2.6 on rpi3 as a micro feeder works fine.

wiedehopf, 2025-02-02:
That's really not enough info to really tackle the issue.

First of all it was an upgrade or the 2.2.6 image in a VM that dies?

tpvainio, 2025-02-02:
It was originally installed from adsb-im-x86-64-native-v2.2.2.iso and upgraded using the GUI button after that day though every stable in between.  After 2.2.6 upgrade it did not ever send anything to fr24 or piaware or at least their statistics say so.  After a few minutes the VM does not respond to the ping anymore.  I did a reinstall using adsb-im-x86-64-native-v2.2.5.iso and restored the backup.

wiedehopf, 2025-02-02:
Well this can't really be debugged according to your description.
There is no reason why the update would mean that you can't ping it anymore.

Without further info, there is nothing to fix.
Guess this issue can stay open for a week in case someone else has a similar issue, after that i'll close it.

tpvainio, 2025-02-02:
Next I'll try to install clean 2.2.6 the same way and monitor that.  After it I'll upgrade the current 2.2.5 to 2.2.6 if it fails again.

wiedehopf, 2025-02-02:
If you want to debug it further that's great of course.
Consider setting the root password via the webinterface so you can log onto the machine from the VM console.
Or are you running the VM headless? I suppose that would be the way to go usually.

tpvainio, 2025-02-02:
Both VMs have been up without any issues for the last two hours.  Nothing breaks so it was some kind of glitch yesterday.  

This is only an anomaly in the logs which is seen 150 times every 60 seconds.

WARNING: list _ADSBIM_STATE_FEEDER_RADARBOX_SN_KEY has length: 1 instead of expected 2 - log: True
WARNING: list FEEDER_MAX_RANGE has length: 1 instead of expected 2 - log: True
WARNING: found list with inconsistent lengths - log: True

closed 2025-02-04 by tpvainio

wiedehopf, 2025-02-04:
As soon as a station is added or removed on a stage2, the anomaly as you call it should stop.

Really i'm not sure why it's not just fixed when it's being logged.