dirkhh/adsb-feeder-image #282: Secure Image Mode, doesn't disable password auth

ScriptEchos, 2025-04-24:
Hi, I'm running v2.3.4 with the raspberrypi64 pi os image. I've added my ssh key, enabled secure image mode and connected using my key but noticed that password authentication is not disabled in the sshd_config.

I'm unsure if you're working this another way but should enabling secure mode set 'PasswordAuthentication no' in the sshd_config?

Very minor side issue, you can add the same ssh key infinite times. Confirmation it already exists or just deduplication might be nice.

dirkhh, 2025-04-24:
as is pointed out in a few places, this isn't intended as a general purpose OS and logging in is intended as a "method of last resort". Since a lot of users don't have the knowledge / skillset to deal with public keys and ssh, both password and public key are enabled.
And yes, you can add the same key infinite times. If that's what you would like to do, go ahead.
From a functionality perspective I'd state that both items fall squarely into the "works as designed" category - but it should be easy enough to disable adding a key that's already there, so I'll look into that.

ScriptEchos, 2025-04-24:
Thanks for the quick response. I guess the first one is more of a UX thing, the implication of secure mode is that I completely disables password based auth.

Key deduplication is super minor but adding a check could help prevent any unexpected behaviour or infinite addition loops down the line.

wiedehopf, 2025-04-25:
So you're saying this text needs changing:
> Make sure you have an SSH key set up and tested before doing this, or you will permanently lock yourself out of this image.
to?
> Make sure you have an SSH key or root password set up and tested before doing this, or you will permanently lock yourself out of this image.

wiedehopf, 2025-04-25:
A random root password is really not a security concern in the grand scheme of things.
If you're worried about that, don't use this image, the webinterface doesn't have known issues but the very nature of how it works isn't the most secure thing. It's after all a python web application.
Thus this text:
> Attempt to make it somewhat harder for someone on the local network to gain access to the image.

wiedehopf, 2025-04-25:
And as for SSH key duplicates.
You add a key and then test if you can log in using the key.

If you add it 100 times until you manage to log in, that's not an issue, the ssh server doesn't work worse because of it.
If you're gonna look into the file of authed keys yourself and don't like it, you're welcome to clean up the file.

wiedehopf, 2025-04-25:
```diff
-            protection attempted here. Make sure you have an SSH key set up
-            and tested before doing this, or you will permanently lock
-            yourself out of this image.
+            protection attempted here. Make sure you have an SSH key or
+            password set up and tested before doing this, or you will
+            permanently lock yourself out of this image.
```
That change in wording should work to clarify this?

wiedehopf, 2025-04-25:
Added something that confirms to the user that the key has been added.
That should be better UI like this?
We will see how it is :)

ScriptEchos, 2025-04-25:
Yes that would be better.

The original version strongly hinted password auth would be completely disabled, but then again a novice user might not make that assumption.

closed 2025-04-25 by ScriptEchos