dirkhh/adsb-feeder-image #285: Reduction in message rate after conversion to Nano feeder

mdkendall, 2025-05-04:
I was running adsb-feeder-image on a Raspberry Pi Zero 2W, feeding a couple of aggregators. All was well.

I created a new "Stage 2" system (on a VM, on Proxmox), connected it to my existing feeder (doing the full import), and then converted the existing feeder to a "Nano" feeder. All appears to work, but after a while it is apparent that the message rate has reduced considerably.

[image]

The number of aircraft in view remains the same, but the messages per aircraft per second has about halved. Why might this be? The SDR gain did change following the reconfiguration, but after a few minutes it settled back to about where it was, and the message rate has not changed.

wiedehopf, 2025-05-04:
Uncheck the bandwidth reduction.

It doesn't hurt anything though.

closed 2025-05-04 by wiedehopf

mdkendall, 2025-05-04:
Well that was easier than I was expecting! Thank you. I was busy looking for problems in my firewall (the two systems are on separate VLANs).