dirkhh/adsb-feeder-image #287: add support gps

ernieek, 2025-05-14:
Adding support for gps (gpsd)

i know for example flightaware can use gpsd date for location setting.

IS it possible to add?

dirkhh, 2025-05-14:
So this is a good summary how to install gpsd and what options to set: https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder/blob/main/README.md#basic-installation-and-configuration-of-your-gps-hardware-and-gpsd-drivers

Once you do that, there's an option on the expert page that allows you to enable that (the option will only show up if you have a working / running gpsd)

Having written all this - in most cases that's actually a step backwards. There is some value if your station is on an RV and you move every few days, but if you are somewhere fixed, there's mostly downsides (because of gps jitter).

wiedehopf, 2025-05-14:
There is zero benefit to gpsd if the receiver is stationary by the way.
It is likely to cause issues rather than be useful in that case.

closed 2025-05-14 by dirkhh

ernieek, 2025-05-14:
I have the option on the expert page. I have enable it, cgps shows a fix 2d and 3d. But flightaware there is no gps tab. Do i forget something?

ernieek, 2025-05-14:
Mmm damm...

Its a mobile station... with a lot of location changes...

Did you try to run gpsd in the docker container connected to the host gpsd?

dirkhh, 2025-05-14:
you'd need a gpsd proxy, right? And I'm not aware of that existing.

wiedehopf, 2025-05-14:
you could try and have the piaware connect to the host.
but no nothing like that has been tried, at least for me it's not a priority.

or wait, does gpsd only connect a single incoming connection?

ernieek, 2025-05-14:
Meaby this wil work om the flightaware container in gpsd

sgpsd -n tcp://ip:port

In stead of /dev/ttyACM0

ernieek, 2025-05-14:
And it seems in piaware config the following is missing:

piaware-config use-gpsd yes

dirkhh, 2025-06-08:
it's possible this got broken somehow - certainly no intentional change. You may be the only person actively trying to use that feature.

Looking at the code there's nothing that I can see that would change the level of support we had in the last release.
Can you say more what changed for you?

CzarofAK, 2025-06-22:
I have the a proxmox with on dedicated GPSD VM which is feeding multiple other VMs and LXCs.
No i tried to implement this to the ADSB Feeder.
I have installed GPSD on the ADSB feeder VM, and setup to pull data from the GPSD server. This seems to work.
But i do not get the GPSD settings enabled in the GUI (expert tab).

wiedehopf, 2025-06-22:
As linked a bit above, you need to make gpsd accessible from the docker network.
Easiest is to just listen on all interfaces modifying gpsd socket as described in the linked ultrafeeder readme.