dirkhh/adsb-feeder-image #288: Can't connect to FlightAware

klubi, 2025-05-15:
Hi,

I believe this is a followup on #246.

Today I tried setting up adsb feeder on a dietpi.
FlightRadar24 login worked just fine, FlightAware did not.

I attached my log, where you can see `temp_container` to start, and then fail with
```
docker run --rm ghcr.io/sdr-enthusiasts/docker-piaware:latest-build-603 received a timeout error after 45.0 with output b'INFO: no specific FEEDER_ID set, falling back to dynamic generated (/var/cache/piaware)\nWARNING: TZ environment variable not set\nallow-auto-updates "no"\nallow-manual-updates "no"\nallow-mlat "yes"\nmlat-results "yes"\nreceiver-type "rtlsdr"\nmlat-results-format "beast,listen,30105 ext_basestation,listen,30106 beast,connect,localhost:30104 "\n[2025-05-15 09:03:11.959][skyaware] 2025-05-15 09:03:11: (server.c.1704) server started (lighttpd/1.4.69)\n[2025-05-15 09:03:13.881][piaware] ****************************************************\n[2025-05-15 09:03:13.881][piaware] piaware version 10.0 is running, process ID 185\n[2025-05-15 09:03:13.910][piaware] your system info is: Linux efc79810c594 6.12.25+rpt-rpi-v7 #1 SMP Raspbian 1:6.12.25-1+rpt1 (2025-04-30) armv7l GNU/Linux\n[2025-05-15 09:03:15.423][piaware] Connecting to FlightAware adept server at piaware.flightaware.com/1200\n[2025-05-15 09:03:17.095][piaware] no ADS-B data program seen listening on port 30005 for 4 seconds, next check in 60s\n[2025-05-15 09:03:17.095][piaware] UAT support disabled by local configuration setting: uat-receiver-type\n[2025-05-15 09:03:30.898][piaware] Connection attempt with adept server at piaware.flightaware.com/1200 timed out\n[2025-05-15 09:03:30.899][piaware] reconnecting in 5 seconds...\n[2025-05-15 09:03:35.901][piaware] Connecting to FlightAware adept server at piaware.flightaware.com/1200\n[2025-05-15 09:03:48.433][piaware] 0 msgs recv\'d from dump1090; 0 msgs sent to FlightAware\n[2025-05-15 09:03:51.043][piaware] Connection attempt with adept server at piaware.flightaware.com/1200 timed out\n[2025-05-15 09:03:51.043][piaware] reconnecting in 5 seconds...\n'
```

I'm running this on RPi B2+, It's not very powerful, however I was monitoring resource usage during login attempt, and neither CPU nor Memory reached 80%, so I'm not sure what's happening here :/

[adsb-feeder-config-KTW-2025-05-15T11-15-04.txt](https://github.com/user-attachments/files/20225362/adsb-feeder-config-KTW-2025-05-15T11-15-04.txt)

wiedehopf, 2025-05-15:
```
May 15 11:12:18 HOSTNAME tailscaled[304]: netcheck: netcheck: UDP is blocked, trying HTTPS
May 15 11:12:18 HOSTNAME tailscaled[304]: netcheck: UDP is blocked, trying ICMP
```
This smells like outbound connection blocking or something like that.
So i'd point to that as a further indicator that the networking is broken.

klubi, 2025-05-15:
Huh... I was just made aware that both 8099 and 1194 ports are blocked for UDP.
Once enabled, all started working just fine.

closed 2025-05-15 by wiedehopf