dirkhh/adsb-feeder-image #289: add a port forwarder

jlmaqui, 2025-05-19:
Hi !

 It should be interested to add a port forwarder like socat on the options, to fwd the ports to the uploader software in other machines, withouth redirecting ports on router, or using it throught cgnat / mobile inet conection; you need a Virtual Radar Server with a public IP,  it´s an easy way to join several receivers on the same location.

VRS works fine on linux with mono ( arm and intel ).

Regards.

dirkhh, 2025-05-19:
that's why Zerotier and Tailscale are supported. That is a FAR better way to connect remote systems and doesn't require to open ports on a router, having a public IP, etc.

closed 2025-05-19 by dirkhh

jlmaqui, 2025-05-19:
Hi!

Yes, nice tools, but the direct option with the vpn through other network / vpn systems is useful too, and on some cases socat is the solution with or not openvpn.

Opening a specific rare port or having a pi on a dmz to do that :-).

Regards.

dirkhh, 2025-05-19:
And anyone is of course able to do that if they so chose.
It's just not something that this project will support.