dirkhh/adsb-feeder-image #305: Incorrect Status Icon Colours

Hamster45105, 2025-06-28:
Sometimes, the new icons do some weird things with the colours, where they are not the colour that they should be. I'm not sure if it's something to do with the code, or just my slow Raspberry Pi not being able to keep up. It usually lasts only 5 - 10 seconds before it fixes itself, but I managed to get some screenshots of when it has occurred:

### Image 1
[image]

### Image 2
[image]

As you can see, the main status icons are all grey and all of the dots are different colours from what they should be. Also, sometimes some of the dots appear black while other appear dark grey.

dirkhh, 2025-06-28:
It takes a while to get a status from the pi. 
The browser does asynchronous calls to get that data and so sometimes it will show things in a default cookie

wiedehopf, 2025-06-28:
I don't think that's it.

It must be an issue with the icon loading code.
Because each icon has an associated color.
So when the icon is shown, the color should match.

In theory we could cut out a lot of the icon loading code if we are fine relying on the cachebust script to just stash all the svgs in iconCache.js
But that's unrelated to this issue.

Hamster45105, 2025-06-28:
Just tested and it looks like it's fixed!

commit 8124b9c2 (2025-06-28): update release notes

and trigger GitHub:
Fixes: #305