dirkhh/adsb-feeder-image #319: Pocket mariner wrong URL

dirkhh, 2025-07-04:


StephanStS, 2025-08-19:
I dealt with this issue in the last days and contacted Steve from the support@pocketmariner.com.
He helped me very kindly and mentioned this:

_BoatBeaconapp.com port 5322 is the general worldwide port for folks to share to. So it should stay as that._ 
_Some folks like to have individual ports assigned to them so they can view the data from their aerial and we 
are happy to do that, we use several different servers for hosting the individual ports._

In my case, the server I have to share is named "crowdais.com" (IP address 54.204.25.151) with my "personal" port.

For the ADS-B feeder application I therefore propose these setting options:
1. Anynomous sharing
   - Then no further entries are needed
   - The feeder application uses  **boatbeaconapp.com:5322**
4. User related port
    - Then the user gets input fields "Sharing host / IP address" and "Personal sharing port number"
    - The feeder application uses these set values

I tested my personal values by patching the `aiscatcher` script in the ADS-B overlay2 directory structure. I there changed the part `boatbeacon.com:$BOATBEACON_UDP_PORT` to `crowdais.com:$BOATBEACON_UDP_PORT`.

```
# BoatBeacon aka Pocket Mariner
if [[ -n "$BOATBEACON_UDP_PORT" ]]; then
    if ! grep -q ":"  <<< "$BOATBEACON_UDP_PORT"; then BOATBEACON_UDP_PORT="crowdais.com:$BOATBEACON_UDP_PORT"; fi
    aiscatcher_command+=("-u ${BOATBEACON_UDP_PORT/:/ }")
```

With this, my personal feeding port on crowdais.com runs well:

[image]

```
```
Also the overall feeder data can be seen on <https://boatbeaconapp.com/ais/station/5322> (but it is quite slow):

[image]

wiedehopf, 2025-08-19:
> is there no env var you can set?
>    2. User related port

there is an env var
https://github.com/sdr-enthusiasts/docker-shipfeeder?tab=readme-ov-file#easy-sharing-with-other-services

Setup - expert - environment variables
Put it there

StephanStS, 2025-08-19:
> there is an env var
> https://github.com/sdr-enthusiasts/docker-shipfeeder?tab=readme-ov-file#easy-sharing-with-other-services
> 
> Setup - expert - environment variables
> Put it there
> […](#)

@wiedehopf : Seems to work for my sharing server (which hostname I was given by Pocket Mariner) using
`BOATBEACON_UDP_PORT="crowdais.com:$BOATBEACON_UDP_PORT"`
Thx for the hint.