dirkhh/adsb-feeder-image #330: FlightRadar24/fr24feed doesn't work when host has IPv6

daviessm, 2025-08-10:
Ref. https://adsblol.zulipchat.com/#narrow/channel/390973-request-for-help/topic/New.20installation.20FlightRadar24.20not.20connecting

When the host is IPv6 enabled, the FlightRadar24 Docker container tries to use IPv6 to resolve DNS, but IPv6 doesn't work in Docker so name resolutions fail and fr24feed doesn't connect to the FlightRadar24 servers.

wiedehopf, 2025-08-10:
This description is misleading

It works in basically all installs even with Ipv6 enabled.
Not sure why it's not working for you.

daviessm, 2025-08-10:
I'm happy to change the description, what troubleshooting steps do you need to make it better?

By default the Docker container's resolv.conf contains nameservers 127.0.0.11 and my router's IPv6 address. As such it can't resolve external addresses. Removing the IPv6 address doesn't make a difference, replacing it with the router's IPv4 address makes name resolution slow but working.

I don't know enough about Docker to know where else to troubleshoot.

wiedehopf, 2025-08-10:
127.0.0.11 is dockers embedded DNS and it should be resolving non-local addresses as well.
It simply acts as a proxy.

Quite possible that you're on an old docker version as well.

You can just exec into the container and test btw
```
docker exec -it fr24feed bash
```

You'll see that even with defaults, you'll be able to nslookup / dig / ping domains on the internet just fine.
If you can't that's just some random bug with docker embedded DNS.

daviessm, 2025-08-10:
> You'll see that even with defaults, you'll be able to nslookup / dig / ping domains on the internet just fine.
> If you can't that's just some random bug with docker embedded DNS.

That was the problem - I couldn't resolve DNS from inside the container, except for the other Docker container names (e.g. uf_1). The queries timed out. So yeah, it seems like the problem is with Docker embedded DNS or its configuration.

daviessm, 2025-08-10:
Yep, this one: `curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash` on Debian 12.

wiedehopf, 2025-08-12:
Oh ok.
You can go back to stable with the button on the homepage.

I just need to fix the capitalization probably.

wiedehopf, 2025-08-12:
Oh you can't have an env var in there ... at least in the older docker compose it seems.

```yml
enable_ipv6: ${AF_DOCKER_IPV6}

image: ${DOZZLE_CONTAINER}
```
interestingly the env var works fine in the image line.

wiedehopf, 2025-08-12:
Ok this seems impossible with the old compose, i'd have to manipulate the yml itself instead of using env vars.
Bit ridiculous.

Might just cancel this option alltogether.
will see.