dirkhh/adsb-feeder-image #331: Script not recognizing docker-compose.

MojoMC, 2025-08-16:
I already installed and used ADSB.IM via the script on Ubuntu Server 24.04 LTS appr. 3 months ago. 
After I installed the current Ubuntu Server LTS version (minimal version to get the size down, only 8 GB eMMC available) I encountered the following problems when running the script. 

` curl https://raw.githubusercontent.com/dirkhh/adsb-feeder-image/main/src/tools/app-install.sh | sudo bash`
leads to
```
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  7169  100  7169    0     0  26156      0 --:--:-- --:--:-- --:--:-- 26356
You appear to be on a debian-style distribution
Please install the missing packages before re-running this script:
apt-get install -y docker-compose
```

But docker-compose is already installed:
```
sudo apt-get install -y docker-compose
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
docker-compose is already the newest version (1.29.2-6ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Running the script again I get the same complaint, docker-compose is not recognized.
I downloaded the script, could not understand why the docker-compose version check failed, commented this part out and ran the script. It finished without error.

However, when I import my ADSB.IM-backup it runs into problems like `ModuleNotFoundError: No module named 'distutils'`.
Strange, since diskutils was removed in Python 3.12.0 in October 2023, as far as I know. I would not expect to depend ADSB.IM on diskutils, especially since I used the install script 3 months ago on Ubuntu Server already without problems.

The system right now with the problems is:
```
cat /proc/version
Linux version 6.8.0-71-generic (buildd@lcy02-amd64-053) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #71-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 22 16:52:38 UTC 2025
```

If you need any other information (logs or whatever), I am happy to provide them.

wiedehopf, 2025-08-16:
System -> share diagnostics

wiedehopf, 2025-08-16:
```
docker compose version
docker-compose version
```
run those, what is the output?

wiedehopf, 2025-08-16:
So your distribution has no working docker compose.
That's really not an adsb.im problem.

You could just use this though: https://get.docker.com/

closed 2025-08-16 by wiedehopf

MojoMC, 2025-08-16:
A functioning workaround for the requirement of the deprecated distutils (removed from Python 3.12.0 in October 2023) was `sudo apt install python3-setuptools` - at least it worked in my case.

Running `docker-compose version` afterwards resulted in
```
docker-compose version 1.29.2, build unknown
docker-py version: 5.0.3
CPython version: 3.12.3
OpenSSL version: OpenSSL 3.0.13 30 Jan 2024
```