dirkhh/adsb-feeder-image #336: Unable to start after a fresh flashing

f5bcb, 2025-09-01:
Hi,
I've been trying for two hours to boot the image (adsb-im-raspberrypi64-pi-2-3-4-5-v3.0.1.img.xz ) on a Raspberry Pi2-B. 
The image has been flashed several times on 32 and 16 GB cards using the rpi-imager or balenaEtcher tools without success: the Raspberry Pi does not start even after 10 minutes. Wan access speed is 1Gb/s on fiber.
The Ethernet network is in place and a WiFi dongle is installed on a USB port.
The symptom is the disk flashing 7 times, then a pause, then flashing 7 times again.
In addition, a network analysis (Advanced IP Scanner) shows that no IP has been assigned to the RPi network port during startup.
I tried the latest Beta version but with exactly the same result.
The Raspberry Pi is not the problem, as I can boot Debian with the Openwebsdr+ application.
So, I don't know what to do in this case.

Thanks for advice

wiedehopf, 2025-09-01:
Try the dietpi image.

raspbian 64bit apparently isn't compatible with the pi2 according to their
download page.
maybe that needs some readme / naming adjustments.

wiedehopf, 2025-09-01:
If it's a 1.1 version PCB you might be out of luck though.

f5bcb, 2025-09-01:
Hi,
Indeed, it's doubly impossible for it to work. I'll look for another application, perhaps ADSB-Exchange, or get myself a newer Pi.
Jeff